pimoroni/vl53l1x-python #3: (install) command 'arm-linux-gnueabihf-gcc' failed with exit status 1

ash73, 2018-07-10:
Hi, I cloned the git repository to my rPi2 and entered:
sudo python setup.py install

After doing some work it generates the error:
command 'arm-linux-gnueabihf-gcc' failed with exit status 1

The full output is:
running install
running bdist_egg
running egg_info
creating python/VL53L1X.egg-info
writing python/VL53L1X.egg-info/PKG-INFO
writing top-level names to python/VL53L1X.egg-info/top_level.txt
writing dependency_links to python/VL53L1X.egg-info/dependency_links.txt
writing manifest file 'python/VL53L1X.egg-info/SOURCES.txt'
reading manifest file 'python/VL53L1X.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'python/VL53L1X.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
copying python/VL53L1X.py -> build/lib.linux-armv7l-2.7
running build_ext
building 'vl53l1x_python' extension
creating build/temp.linux-armv7l-2.7
creating build/temp.linux-armv7l-2.7/api
creating build/temp.linux-armv7l-2.7/api/core
creating build/temp.linux-armv7l-2.7/api/platform
creating build/temp.linux-armv7l-2.7/python_lib
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_api_calibration.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_api_calibration.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_core.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_core.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_core_support.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_core_support.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_api_core.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_api_core.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_api_preset_modes.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_api_preset_modes.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_silicon_core.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_silicon_core.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_register_funcs.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_register_funcs.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_wait.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_wait.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_error_strings.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_error_strings.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_api_strings.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_api_strings.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/core/vl53l1_api.c -o build/temp.linux-armv7l-2.7/api/core/vl53l1_api.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I. -Iapi/core -Iapi/platform -I/usr/include/python2.7 -c api/platform/vl53l1_platform.c -o build/temp.linux-armv7l-2.7/api/platform/vl53l1_platform.o
api/platform/vl53l1_platform.c:279:1: warning: multi-line comment [-Wcomment]
 //#define trace_print(level, ...) \
 ^
api/platform/vl53l1_platform.c:283:1: warning: multi-line comment [-Wcomment]
 //#define trace_i2c(...) \
 ^
api/platform/vl53l1_platform.c: In function ‘VL53L1_WaitValueMaskEx’:
api/platform/vl53l1_platform.c:321:2: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
  for(int32_t x = 0; x < attempts; x++){
  ^
api/platform/vl53l1_platform.c:321:2: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

Please help.

urbie-mk2, 2018-07-11:
try 
export CXXFLAGS="-std=c++11"
export CFLAGS="-std=c99"

ash73, 2018-07-11:
Still fails with the same error :(

Gadgetoid, 2018-07-13:
You could try adding it to the Makefile, IE:

```makefile
.PHONY: clean lib

SRC = api
LIB = libvl53l1_api.a
LDFLAGS :=
CFLAGS := -Iapi/core -Iapi/platform -std=c99

SRC_FILES := $(wildcard $(SRC)/*/*.c)
SRC_FILES := $(wildcard $(SRC)/*/*.c)
OBJ_FILES := $(patsubst $(SRC)/*/*.c,$(SRC)/*/%.o,$(SRC_FILES))

$(LIB): $(OBJ_FILES)
	$(CC) $(LDFLAGS) $(CFLAGS) -fPIC -shared -o $@ $^

$(SRC)/%.o: $(SRC)/%.c
	$(CC) $(LDFLAGS) $(CFLAGS) -c -o $@ $<

test: test.c
	$(CC) $(CFLAGS) $(LDFLAGS) -Lvl53l1_api -o test test.c
```

ash73, 2018-07-13:
Still fails with the same error. Am I missing some dependencies?

Gadgetoid, 2018-07-13:
You could possibly cheat and edit `api/platform/vl53l1_platform.c` to remove the declaration outside of the for loop.

```
int32_t x;
for(x = 0; x < attempts; x++){
```

You're not missing a dependency, just my Makefile isn't particularly friendly to your version of GCC (I'm using 6.3.0-18+rpi1+deb9u1).

Gadgetoid, 2018-07-13:
Curious, that change to the Makefile gets me building with `gcc-4.4`, sure you're changing it in the right place?

ash73, 2018-07-13:
I copied and pasted your text into the file "Makefile" in the folder vl53l1x-python (the other files in that folder are MANIFEST.in setup.py test.c and test.py), I think the only bit that changed was adding -std=c99 to the end of the line starting CFLAGS :=

I looked at api/platform/vl53l1_platform.c but the declaration is different in mine:

	int32_t attempts = timeout_ms / poll_delay_ms;
	for(int32_t x = 0; x < attempts; x++){

Not sure what to change / remove.

ash73, 2018-07-13:
p.s. gcc --version

gcc (Raspbian 4.9.2-10+deb8u1) 4.9.2

urbie-mk2, 2018-07-16:
@ash73 
Could you create a clean install of your operating system, then compile again and then compare the results ? If it isn't the compiler, maybe it is some modification in the operating system that you are not aware of.

Gadgetoid, 2018-07-17:
@ash73 this time it has built successfully. The `usleep` error is a warning and not a fatal compilation error. Granted I'm not sure *why* it's undeclared and I need to fix that, but the code should work.

If your C compiler is playing ball now you should run:

```
python setup.py build
```

Gadgetoid, 2018-07-18:
That's just a normal side-effect of my lazy programming ;)

You should be able to `sudo python setup.py install` to install the library and start working from there.

That said, after all this if you're intending to use it with Python on a Raspberry Pi for example you can just "sudo pip install VL53L1X"

ash73, 2018-07-19:
Thanks very much for your help. I think the main problem was a corrupted Raspbian install on my side, sorry to waste your time. Anyhow it's all working now, thanks again.

closed 2018-07-19 by ash73