pimoroni/vl53l1x-python #6: Multiple sensor and address change

liningwei, 2018-07-24:
Hi,

I'm doing a project where multiple of VL53L1Xs are needed and they share the same bus. I installed the library and did some testing. 

The function change_address(addr) works in a way that I can see the physical i2c address being changed on raspberry pi 'i2cdetect -y 1', but it can't range anymore. It gives 'remote I/O error'.

It works well when I do
s1=VL53L1X() #0x29
s1.open()
s1.start_ranging()
s1.get_distance()

but once I put in change_address(addr)
s1=VL53L1X() #0x29
s1.open()
s1.change_address(0x30) #update address to allow more sensors
s1.start_ranging()
s1.get_distance()

Any idea why it'd return I2C I/O error? 

-----------------------------------------------------------
Also in VL53L1X.py, inside change_address(addr)
    def change_address(self, new_address):
            _TOF_LIBRARY.setDeviceAddress(self._dev, new_address)

The instance variable sensor.i2c_address is never updated so I added
           self.i2c_address = new_address

but it didn't fix the issue with I2C I/O error

liningwei, 2018-07-24:
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 315, in 'calling callback function'
  File "/home/ubuntu/.local/lib/python2.7/site-packages/VL53L1X.py", line 91, in _i2c_read
    self._i2c.i2c_rdwr(msg_w, msg_r)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/smbus2/smbus2.py", line 483, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
IOError: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 315, in 'calling callback function'
  File "/home/ubuntu/.local/lib/python2.7/site-packages/VL53L1X.py", line 109, in _i2c_write
    self._i2c.i2c_rdwr(msg_w)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/smbus2/smbus2.py", line 483, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
IOError: [Errno 121] Remote I/O error

This is the error when I run s1.get_distance() after changing address on it

Gadgetoid, 2018-10-15:
This is pretty much where I got with implementing sensor address change. I think there's something fundamental we're overlooking, but it's one of those things that I haven't had time to look into since.

The confusing use of 7/8 bit i2c addresses in various places didn't help, iirc!

NagyAttila, 2018-10-25:
Alternatively, you can use multiple sensors with the same default i2c address (0x29) having only turn on one at the same time.

Something like this:
`import VL53L1X
from time import sleep
import time
import RPi.GPIO as GPIO

SHUTX_PIN_1 = 20
SHUTX_PIN_2 = 16

def get_and_print_measurement():
    # Start ranging, 1 = Short Range, 2 = Medium Range, 3 = Long Range
    start = time.time()
    tof.start_ranging(3)
    distance_in_mm = tof.get_distance()
    tof.stop_ranging()
    print("sensor on pin: %d\tvalue: %d\ttime: %f" % (pin, distance_in_mm,time.time()-start) )

def toggle_pin(pin):
    if pin == SHUTX_PIN_2:
        new_pin = SHUTX_PIN_1
    else:
        new_pin = SHUTX_PIN_2
    return new_pin

GPIO.setwarnings(False)

# Setup GPIO for shutdown pins on each VL53L0X
GPIO.setmode(GPIO.BCM)
GPIO.setup(SHUTX_PIN_1, GPIO.OUT)
GPIO.setup(SHUTX_PIN_2, GPIO.OUT)

# Set all shutdown pins low to turn off each VL53L0X
GPIO.output(SHUTX_PIN_1, GPIO.LOW)
GPIO.output(SHUTX_PIN_2, GPIO.LOW)
sleep(1)

# Start with first sensor
pin = SHUTX_PIN_1
GPIO.output(pin, GPIO.HIGH)

# Initialise the i2c bus and configure the sensor
tof = VL53L1X.VL53L1X()
tof.open()

while True:
    # start = time.time()
    pin = toggle_pin(pin)
    GPIO.output(pin, GPIO.HIGH)
    get_and_print_measurement()
    GPIO.output(pin, GPIO.LOW)
    # print("Elapsed time: %f sec" % (time.time()-start))
    sleep(0.05)
`

Gadgetoid, 2019-10-31:
I believe this issue is a duplicate of, or supplemental to:
https://github.com/pimoroni/vl53l1x-python/issues/13
https://github.com/pimoroni/vl53l1x-python/issues/14

I'm closing it- but this mention should keep a link in the above issues for reference. Please use above issues for discussion.

closed 2019-10-31 by Gadgetoid