pimoroni/vl53l1x-python #7: No such device or address Error

jimclydegm, 2018-08-14:
I'm getting a "No such device or address" error whenever i build and run in Ubuntu 16.04 LTS. 
Is it possible to run this program in linux systems?

I tried to change the bus instance from
`tof = VL53L1X.VL53L1X(i2c_bus=1, i2c_address=0x29)`
to
`tof = VL53L1X.VL53L1X(i2c_bus=0, i2c_address=0x29)`
but no luck.

Here's the error trace

> Traceback (most recent call last):
>   File "_ctypes/callbacks.c", line 315, in 'calling callback function'
>   File "/usr/local/lib/python2.7/dist-packages/VL53L1X-0.0.1-py2.7-linux-x86_64.egg/VL53L1X.py", line 94, in _i2c_read
>     self._i2c.i2c_rdwr(msg_w, msg_r)
>   File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 483, in i2c_rdwr
>     ioctl(self.fd, I2C_RDWR, ioctl_data)
> IOError: [Errno 6] No such device or address

Gadgetoid, 2018-08-15:
What do you see if you run:

```
sudo i2cdetect -y 1
```

and/or

```
sudo i2cdetect -y 0
```

sandyjmacdonald, 2018-08-15:
Just checking... you've soldered the header onto the breakout?

jimclydegm, 2018-08-16:
@Gadgetoid both commands return the following results
`root@xxxxxxxxxx:~# i2cdetect –y 1
    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f 
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- `

I suppose this means that the sensor is not recognized?
This no longer your concern but could you tell me how to get it recognized in Ubuntu? 
It works well in windows, I use sensorviewer from Intel

jimclydegm, 2018-08-16:
@sandyjmacdonald  This is my current [setup](https://drive.google.com/file/d/19d_MRyCPFYtlrK8loC-4sHvAeLKOUssW/view?usp=sharing)

Gadgetoid, 2018-10-15:
Ubuntu/Windows? What platform are you running this on? I'm afraid I may not be much help regarding anything but Pi/Raspbian since that's what I ported this library to support. I'm not averse to helping where I can on other platforms, though, but my knowledge may fail me!

Gadgetoid, 2018-10-30:
Any luck?

closed 2019-10-31 by Gadgetoid