pimoroni/vl53l1x-python #9: install via breakout garden fails

koaning, 2018-10-06:
for this error after installing.

```
pi@raspberrypi:~/breakout-garden-master $ sudo ./install.sh
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

Breakout Garden requires I2C. We've enabled it for you.

Breakout Garden: Installer. (2 breakout(s) found)

vl53l1x time of flight:         Installed
bme680 weather sensor:          Installed

Installing 2 module(s). Enter to continue (Ctrl+C to cancel)...
pi@raspberrypi:~/breakout-garden-master $
pi@raspberrypi:~/breakout-garden-master $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import VL53L1X
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/VL53L1X.py", line 25, in <module>
    from smbus2 import SMBus, i2c_msg
ImportError: No module named smbus2
```

koaning, 2018-10-06:
It was fixed with a simple: 

```
pip install smbus2
```

Gadgetoid, 2018-10-15:
Thanks for reporting this. Looks like I'd flubbed the `setup.py` file here, and forgot to properly depend upon `smbus2`. I've pushed a fix. I still need to publish the new library version, though.

closed 2019-10-31 by Gadgetoid