pimoroni/vl53l1x-python #10: Tof.open() gets stuck

lilyanne12, 2018-10-20:
Hello,

I'm trying to test the sensor for a robot I'm making. I have successfully imported the package. I run the following:

import VL53L1X

tof = VL53L1X.VL53L1X(i2c_bus=1, i2c_address=0x29)
print("initialised")
tof.open()
print("opened")

However it seems to get stuck on the tof.open() stage and does nothing until I keyboard interrupt the program. There are no error messages so I'm not sure how to debug it!

Any help would be greatly appreciated.

Thanks,
Lily

lilyanne12, 2018-10-21:
Edit: so I've isolated the section that doesn't compete within tof.open(). The first two lines compete then it gets stuck on this one:

self._dev = _TOF_LIBRARY.initialise(self.i2c_address)

Let me know if you have any ideas. 
Thanks,
Lily

Gadgetoid, 2018-10-31:
What do you see if you run `sudo i2cdetect -y 1`- does a device with address `0x29` appear?

lilyanne12, 2018-11-02:
Yes it does

IrisGreen, 2019-07-24:
I have the same problem and I found it's because the vl53l1x was broken

Gadgetoid, 2019-10-31:
I've seen this happen a few times during my testing, occasionally the sensor gets into a bad state and needs unplugging/replugging. This is obviously not good, as any permenant setup would not let you rip out a sensor on a whim to reset it. 

While a dead sensor can *also* cause this problem, I'm keeping this open to track this as a potential bug in the library that needs fixing.

Gadgetoid, 2020-08-03:
It's been nearly two years and no additional instances of this issue reported here. Will close for now and hope it is, indeed, hardware and not software at fault. Thanks all!

closed 2020-08-03 by Gadgetoid