pimoroni/vl53l1x-python #11: Python 3.6.6

josemotta, 2018-12-03:
Hi. I´m trying to build a distance sensor using VL53L1X and homeassistant 0.82.1 installed with hassio 140. This version use Python 3.6.6 which is not currently supported by VL53L1X library. Please check details below.

Would it be possible to build the VL53L1X library for Python 3.6.6?

pi@hassio:~ $ docker exec -it bbb bash
bash-4.4# pip install vl53l1x -vvv
…
Collecting vl53l1x

https://pypi.org:443 “GET /simple/vl53l1x/ HTTP/1.1” 304 0
Analyzing links from page https://pypi.org/simple/vl53l1x/
Skipping … /VL53L1X-0.0.1-cp27-cp27mu-linux_armv7l.whl it is not compatible with this Python
Skipping … /VL53L1X-0.0.1-cp34-cp34m-linux_armv7l.whl# it is not compatible with this Python
Skipping … /VL53L1X-0.0.1-cp35-cp35m-linux_armv7l.whl it is not compatible with this Python
Skipping … VL53L1X-0.0.2-cp27-cp27mu-linux_armv7l.whl it is not compatible with this Python
Skipping … VL53L1X-0.0.2-cp27-none-linux_armv6l.whl it is not compatible with this Python
Skipping … VL53L1X-0.0.2-cp34-cp34m-linux_armv6l.whl it is not compatible with this Python
Skipping … VL53L1X-0.0.2-cp34-cp34m-linux_armv7l.whl it is not compatible with this Python
Skipping … VL53L1X-0.0.2-cp35-cp35m-linux_armv6l.whl it is not compatible with this Python
Skipping … VL53L1X-0.0.2-cp35-cp35m-linux_armv7l.whl it is not compatible with this Python
Could not find a version that satisfies the requirement vl53l1x (from versions: )

Gadgetoid, 2019-04-15:
Right now I only build Python packages into .whl format for Python versions that are available on Raspbian- Python 2.7, 3.4 and 3.5.

To build for 3.6 you would have to clone this GitHub repository, run:

```
make
python setup.py build
python setup.py install
```

josemotta, 2019-04-15:
Thanks for your reply. I cloned the upgraded version from @NagyAttila and built it for 3.6, please check links below:

https://github.com/josemotta/vl53l1x-python

https://pypi.org/project/VL53L1X2/

closed 2019-10-31 by Gadgetoid