pimoroni/vl53l1x-python #12: Possible duplicate initialisations of i2c

citiusdevel, 2019-01-13:
I intend to use this sensor, but already have i2c communications with other devices within my program. Will the 'open' call recognise that the i2c is already initialised, or will there be a conflict ?

citiusdevel, 2019-01-13:
To clarify a little - I will be using the VL53L1X processing within it's own thread. 
I have done the usual i2c_bus = smbus.SMBus(1), with 'i2c_bus' set up as a global variable.

Gadgetoid, 2019-01-15:
Individual i2c transactions are thread-safe- you cannot interrupt them by starting a transaction from another process- but you should be careful that you don't interfere with a multi-transaction command that might be timing sensitive.

citiusdevel, 2019-01-15:
Thanks for the prompt reply.
The thread-safe situation is good and I appreciate what you mean on time sensitive processing. Having realised this, I might change my theory of a global variable and put the complete i2c code relating to a device into each relevant thread.
I have tested out the different elements of my roving robot program, but it is now time to plug it all together and see what happens. Then the real fun starts.

closed 2019-02-15 by Gadgetoid