pimoroni/vl53l1x-python #13: Using with tca9548a

Eponeshnikov, 2019-02-15:
Can I use this sensors and this lib with tca9548a. I know that vl53l0x lib support tca9548a. How can I use vl53l1x with tca9548a?

Eponeshnikov, 2019-02-15:
I use it on jetson tx2

Gadgetoid, 2019-02-15:
This is not something I’ve tested or tried, so I can’t be much help- sorry. This library is primarily to support our i2c breakout board, and I don’t get much (if any) time to try things like this out.

However- I don’t mind leaving this issue open  in case someone drops by and has an answer for you.

lvermue, 2019-10-27:
@Eponeshnikov @Gadgetoid 
Should work now.

Gadgetoid, 2019-10-30:
Thank you! Very much appreciated.

I'm going to leave this issue open for the time being, both pending feedback from @Eponeshnikov if they are still interested and to remind myself to test/regression test this before the next package release.

closed 2019-10-31 by Gadgetoid