pimoroni/vl53l1x-python #16: Changing the timing budget

dfrEak, 2019-04-08:
I tested in Raspberry Pi W Zero
I want to know is there any way to change the timing budget
because right now it took 0.12 - 0.14 sec per reading
I want to increase the frequency, thank you

Gadgetoid, 2019-04-15:
Looks like the culprit may be here: 

https://github.com/pimoroni/vl53l1x-python/blob/c7116040d7074b3e5962983220b533a7571f1541/python_lib/vl53l1x_python.c#L94-L123

Regardless of the mode requested, it's setting a 136ms total wait period between updates- a 66ms Timing budget plus a 70ms inter-measurement period.

I'm looking into it!

Gadgetoid, 2019-04-15:
I've raised PR #17 to provide the functions you need for setting a custom timing budget.

dfrEak, 2019-04-19:
Thank you gadgetoid
Yeah I found that setting too, but I can not find the c file in my system

dfrEak, 2019-07-01:
I am sorry for my late reply, but I think now it is working
I still trying to improve the frame rate by using 4 sensors
and it seems that even though I reduce the time budget, the frame rate still small (2 or 3 frames per sec)
because I use the switching method
I try to use them simultaneously to increase the frame rate but failed in changing the address (sorry maybe this one is in another topic)

Gadgetoid, 2019-10-31:
I've now merged this pending release- any tests/feedback on the current codebase would be appreciated if you're still working with it.

closed 2019-10-31 by Gadgetoid

dfrEak, 2019-11-26:
I am sorry for my late reply
yeah, I still working with it
I will give you the feedback later