pimoroni/vl53l1x-python #18: IOError: [Errno 121] Remote I/O error

MrThanlon, 2019-05-18:
```
IOError: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 315, in 'calling callback function'
  File "/usr/local/lib/python2.7/dist-packages/VL53L1X.py", line 99, in _i2c_read
    self._i2c.i2c_rdwr(msg_w, msg_r)
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 497, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
IOError: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 315, in 'calling callback function'
  File "/usr/local/lib/python2.7/dist-packages/VL53L1X.py", line 99, in _i2c_read
    self._i2c.i2c_rdwr(msg_w, msg_r)
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 497, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
```

```
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```

Gadgetoid, 2019-05-24:
The VL53L1X I2C address is 0x52 so it's not showing up there at all. How is your sensor connected?

MrThanlon, 2019-05-25:
@Gadgetoid I just connected to I2C pin, SCL and SDA.

IrisGreen, 2019-07-16:
Hi, there! I just run the example/distance.py and got the same problem, 
the i2c address was:
` i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  `
so I was wondering if you know the solution:-)

dfrEak, 2019-09-04:
I think that 0x29 is the VL53L1X I2C address
because it is there, after I connect the sensor

Gadgetoid, 2019-10-31:
@dfrEak you are correct- I have no idea where I got the completely wrong i2c address from.

Closing for now, since I believe it 's a wiring or hardware issue. @MrThanlon please feel free to re-open if necessary.

closed 2019-10-31 by Gadgetoid