pimoroni/vl53l1x-python #20: pip install fails on Raspbian Buster for pi Zero W

chad-ermacora, 2019-07-16:
I have a Raspberry Pi Zero W with Raspbian Buster (Released July 10th, 2019).  When I go to install 'pip3 install vl53l1x', it gives an "ERROR: No matching distribution found for vl53l1x".

It works fine on the Raspberry Pi 3B+ with the exact same version of Raspbian oddly enough.

Gadgetoid, 2019-07-16:
That's odd- I'm shipping a wheel for Python 3 so it should (and does on my Pi 4 running Buster) install.

What do you see if you run:

```
python3 --version
```

chad-ermacora, 2019-07-16:
Python --version = Python 3.7.3, which is the same version that's used on the Pi 3B+, which makes sense since I'm flashing the exact same Raspbian image on both.  The only difference I have found so far is that the Pi Zero W is arm61, where the Pi 3B+ is arm71.  

I have now also re-flashed the image to see if maybe there was an oddity during the flash process, but the same result occurred after the re-flash.

VBaczynski, 2019-10-26:
> The only difference I have found so far is that the Pi Zero W is arm61, where the Pi 3B+ is arm71.

There is no version for arm61:
https://pypi.org/project/VL53L1X/#files

Gadgetoid, 2019-10-31:
Ooof. Perhaps my more eggregious mistake is not releasing a source package, so even `piwheels` can't help here.

I've tagged this with the `0.0.4` milestone to remind myself to push out an arm6l package with the - hopefully imminent - next release.

closed 2019-10-31 by Gadgetoid