pimoroni/vl53l1x-python #21: Non_blocking measurement or interrupt method

farmingmachine, 2019-07-29:
Hey,

it it possible to realize the non-blocking method or the physical interrupt driven method? I tried to change the "wait_method" in the api for measurements to NON_BLOCKING and recompiled, but in python I still need about the Intermeasurement_period duration for finishing the "get_distance" line, which is too time consuming. Any help?

Kind regards,

Malik

Gadgetoid, 2019-08-20:
It's something I can look into, but I can't promise when. As near as I can tell the `wait_method` you mention doesn't actually change the code from BLOCKING to NON_BLOCKING unless you specifically implement a non-blocking version of the wait. This wouldn't translate well to Python I'd wager.

In this case we probably want not to wait at all, but to allow you to do something akin to `isDistanceReady`. Chopping the blocking check on ready state out of the current all-in-one function and letting you call it manually. The vl53l1x API code- which I'm avoiding making any changes to- does not make this easy since everything is buried so many levels deep in unnecessary abstraction but it looks like this is the key:

```
VL53L1_Error VL53L1_WaitMeasurementDataReady(VL53L1_DEV Dev)
{
	VL53L1_Error Status = VL53L1_ERROR_NONE;

	LOG_FUNCTION_START("");

	/* Note that the timeout is given by:
	* VL53L1_RANGE_COMPLETION_POLLING_TIMEOUT_MS defined in def.h
	*/

	Status = VL53L1_poll_for_range_completion(Dev,
			VL53L1_RANGE_COMPLETION_POLLING_TIMEOUT_MS);

	LOG_FUNCTION_END(Status);
	return Status;
}
```

I wonder what effect changing the `VL53L1_RANGE_COMPLETION_POLLING_TIMEOUT_MS` to something like 1ms would have. At first glance it would break `getDistance` because that function doesn't handle a timeout exception (ooof, I should fix that), but if you were to also implement `isDistanceReady` as something like:

```
bool isDistanceReady(VL53L1_Dev_t *dev)
{
    VL53L1_Error Status = VL53L1_ERROR_NONE;
    Status = VL53L1_WaitMeasurementDataReady(dev);
    return Status == VL51L1_ERROR_NONE;
}
```

And then call it before calling `getDistance` to ensure a reading is ready, then it should work.

This is assuming `WaitMeasurementDataReady` doesn't do something silly like trigger the measurement.

This is one of those cases I'd probably handle with a thread, but that often causes more problems than it solves.

Are you trying to interleave ranging measurements with other code?

farmingmachine, 2019-08-21:
Hey!

I also did some changes, which didnt have a nice result, but they might help us narrow down the relevant code:

I used the following changed getDistance function in the “vl53l1x_python.c” file 
````c
int32_t  getDistance(VL53L1_Dev_t *dev)

{

VL53L1_Error Status = VL53L1_ERROR_NONE;

int32_t  current_distance = -1;

uint8_t  dataready = 0;

Status = VL53L1_GetMeasurementDataReady(dev, &dataready);  *// VL53L1_GetMeasurementDataReady takes a pointer, so we pass the address of dataready*

if  (dataready == 1)

{

Status = VL53L1_GetRangingMeasurementData(dev, pRangingMeasurementData);

current_distance = pRangingMeasurementData->RangeMilliMeter;

VL53L1_ClearInterruptAndStartMeasurement(dev);

}

return  current_distance;

}
'''

Then i recompiled the library into python and tried to measure the time spent for the tof.getdistance() line with this code

tof.set_timing(30000, 40)
print("budget", tof.get_timing())
time.sleep(0.5)

while  running:

    elapsed = 0
    current_time = time.time()
    time_before = time.time()
    distance_in_mm = tof.get_distance() # Grab the range in mm
    print("time spend", time.time()-time_before) # print the time spent  **for**  grabbing the range

    if  distance_in_mm>0:
        print("Distance: {}mm".format(distance_in_mm))

    while  elapsed < 0.05: # wait to complete the 50 ms cycle

        elapsed = time.time()-current_time

    print("50 ms loop done"+"\n")

However, the result is still a blocking/waiting communication for when data is ready (dataready=1). I  As you can see in the terminal output below , it is 1 ms for when data is not ready and about 30 ms for when data is ready (tried it with different timing budgets and periods, but same effect of 30 ms waiting)

so it HAS to do sth with the VL53L1_GetRangingMeasurementData() function

I'll also look into your ideas!

Looking frward to hearing from you!

![terminal](https://user-images.githubusercontent.com/50686941/63418259-c180be80-c43d-11e9-9cdb-6299ee54dc57.png)

farmingmachine, 2019-08-23:
Okay, the culprit has been found. With the non-blocking code i posted here, i got it down to 7 ms with 400kHz i2c frequency. Turns out that the 100 kHz default value made it slower before.
We could integrate this non-blocking functional method!

Cheers,

albzn540, 2019-09-11:
Hi @farmingmachine,
Non-issue-relevant question, but I'm also trying to change stuff in the `vl53l1x_python.c` file, but I can't for the love of my life not recompile the c-file into a shared .so file.
Could you explain how you did it?

Cheers

farmingmachine, 2019-09-16:
> Hi @farmingmachine,
> Non-issue-relevant question, but I'm also trying to change stuff in the `vl53l1x_python.c` file, but I can't for the love of my life not recompile the c-file into a shared .so file.
> Could you explain how you did it?
> 
> Or did you simply change the `vl53l1x_python.c` file and then reinstall the package?
> Cheers

Hey @albzn540 ,

Yeah I just changed the code in the relevant files and reinstalled the package with "  pip3 install .  " (after deleting the old module).
Hope this short answer helps.

Kind regards,

Gadgetoid, 2019-10-31:
It's been a while- I don't suppose you have anything worthy of wrangling into a pull-request?

I've just merged a whole suite of updates to the library, including syncing the latest API code from ST so that might put a spanner in your works.

From your comments it looks like implementing an `isDataReady` function could work and your problems with a read time of 30ms are more related to bus speed than any deliberate attempt from the library to block on some expected state?

closed 2023-01-20 by Gadgetoid