pimoroni/vl53l1x-python #23: version 0.0.4 slow and with read problems?

JimVanEeden, 2019-11-04:
Hi, I just built/installed version 0.0.4 myself because I really needed the multiplexer functionality (didn't get that to work yet by the way, but that might be hardware issue). 

Unfortunately, I ran into a problem with this version, even without the multiplexer. So with just `VL53L1X.VL53L1X(i2c_bus=i2c_bus, i2c_address=i2c_address)`. Here's a bit of the sensor output:

```
Distance: 175mm
Distance: -1185mm
Distance: -1185mm
Distance: 178mm
Distance: -1185mm
Distance: 181mm
Distance: -1185mm
Distance: -1185mm
Distance: -1185mm
Distance: 180mm
Distance: -1185mm
Distance: -1185mm
```
The time between these reads is also ~1 second, with a sleep time in the run loop of 0.1. When switching back to version 0.0.3, everything works exactly as expected again.

Just wanted to give a heads up. I'm eagerly awaiting a version that works well and with multiplexer support.

Gadgetoid, 2019-11-04:
A lot has changed between v0.0.3 and v0.0.4 so I'm not surprised something has gone wrong: https://github.com/pimoroni/vl53l1x-python/compare/v0.0.3...v0.0.4

However, I'm at a loss to re-create your issue. Could you post the code you're using to set up and run the sensor? And some idea of how the sensor is wired up?

What happens if you run the `graph.py` example? Since that's giving consistent, and fast readings in my setup.

JimVanEeden, 2019-11-05:
Just tried with freshly installed raspberry pi. This time I didn't build myself because I noticed 0.0.4 is now installable with pip.

The test is ran with a slightly modified version on distance.py in the examples:
```
while running:
    start = time.time()
    distance_in_mm = tof.get_distance()
    print("Distance: {}mm, time: {}".format(distance_in_mm, time.time() - start))
    time.sleep(0.2)
```

I ran: `sudo python3 ~/python/vl53l1x-python/examples/distance.py` with 0.0.4 installed, resulting in the following output: 

```
VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID : 
ProductRevisionMajor : 1
ProductRevisionMinor : 7
Distance: 647mm, time: 0.07596778869628906
Distance: 650mm, time: 0.7849745750427246
Distance: 649mm, time: 0.7818386554718018
Distance: 649mm, time: 0.7822270393371582
Distance: 651mm, time: 0.7816939353942871
Distance: 651mm, time: 0.781696081161499
Distance: 649mm, time: 0.7817797660827637
Distance: 649mm, time: 0.7816405296325684
Distance: 650mm, time: 0.7818257808685303
Distance: 653mm, time: 0.7816331386566162
Distance: 110mm, time: 0.7815234661102295
Distance: 85mm, time: 0.7817280292510986
Distance: 97mm, time: 0.7819225788116455
Distance: 99mm, time: 0.7802751064300537
Distance: 97mm, time: 0.781930685043335
Distance: 642mm, time: 0.7614901065826416
Distance: 649mm, time: 0.8010272979736328
```

Exactly the same command, but with 0.0.3 installed:
```
VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID : 
ProductRevisionMajor : 1
ProductRevisionMinor : 7
Distance: 650mm, time: 0.10051751136779785
Distance: 648mm, time: 0.019917964935302734
Distance: 649mm, time: 0.0198211669921875
Distance: 649mm, time: 0.03726077079772949
Distance: 650mm, time: 0.03724074363708496
Distance: 648mm, time: 0.03722405433654785
Distance: 652mm, time: 0.03720593452453613
Distance: 651mm, time: 0.03722238540649414
Distance: 649mm, time: 0.03721451759338379
Distance: 651mm, time: 0.03720545768737793
Distance: 650mm, time: 0.03724789619445801
Distance: 651mm, time: 0.03725695610046387
Distance: 649mm, time: 0.03719019889831543
Distance: 648mm, time: 0.03726482391357422
Distance: 652mm, time: 0.037218570709228516
Distance: 649mm, time: 0.037195444107055664
Distance: 648mm, time: 0.037178754806518555
Distance: 649mm, time: 0.03721952438354492
```

You can clearly see the big difference in time between the reads. I did not manage to reproduce the -1185 reads however.

Gadgetoid, 2019-11-05:
The slowness I think I can explain, but I had no idea how the read errors factored in. I'm glad they've subsided.

One of the big changes between 0.0.3 and 0.0.4 (in retrospect I should probably have gone for 0.1.0 and I realised that just a little too late) is the addition of the `set_timing` command and some fairly sweeping changes to "Start Ranging" that clearly adversely affect the default timings.

You can set your own timings to get back on track as demonstrated by the `graph.py` example:

```
UPDATE_TIME_MICROS = 1000
INTER_MEASUREMENT_PERIOD_MILLIS = 100
tof.set_timing(UPDATE_TIME_MICROS, INTER_MEASUREMENT_PERIOD_MILLIS)
```

Before these functions existed the library had a hard-coded measurement timing budget of 66000 microseconds and an inter-measurement period of 70 milliseconds so in your case adding these lines should get `distance.py` to match in both 0.0.3 and 0.0.4:

```
UPDATE_TIME_MICROS = 66000
INTER_MEASUREMENT_PERIOD_MILLIS = 70
tof.set_timing(UPDATE_TIME_MICROS, INTER_MEASUREMENT_PERIOD_MILLIS)
```

I believe the command `tof.set_distance_mode(3)` is a short-hand for setting a timing budget (I'm a bit rusty on VL53L1X since I'm juggling so many sensors) and can possibly be ommitted where a timing budget is specified explicitly.

JimVanEeden, 2019-11-05:
Ok thanks so much for your help, this was the answer to fix the problem indeed. 

You think you can maybe add this to the README? Also how to use the multiplexer with `VL53L1X.VL53L1X(i2c_bus=1, i2c_address=0x70, tca9548a_num=1, tca9548a_addr=0x29)`, because I'm also struggling with that a bit ;)

closed 2019-11-05 by JimVanEeden