pimoroni/vl53l1x-python #24: Cannot get the sensor to work with tca9548a multiplexer

JimVanEeden, 2019-11-07:
I've been trying for quite a while to get the sensor to work with the tca9548a multiplexer, which I was glad to see version 0.0.4 supported. Unfortunately I could not get it to work. The sensor works perfectly fine without the multiplexer, and as far as I can see, the hardware setup is working as well. Some tests with a C library confirm this.

Just running the `examples/distance_multiplexer.py` file results in the following error: 

```
Traceback (most recent call last):
  File "distance_multiplexer.py", line 22, in <module>
    tof1 = VL53L1X.VL53L1X(i2c_bus=1, i2c_address=0x29, tca9548a_num=1, tca9548a_addr=0x70)
  File "/usr/local/lib/python3.7/dist-packages/VL53L1X.py", line 88, in __init__
    raise RuntimeError("VL53L1X not found on adddress: {:02x}".format(self.i2c_address))
RuntimeError: VL53L1X not found on adddress: 29
```

Hopefully somebody can help me out here.

Gadgetoid, 2019-11-07:
The code to support this was kindly contributed by @lvermue and I'm afraid I don't currenrtly have a tca9548a handy to verify his tests. A quick check over the code suggests it's correct- have you been able to switch the tca9548a via the command-line and observe the changes on the bus?

This should write a single `0x02` byte to the tca9548a on address `0x70`:

```
i2cset -y 1 0x70 0x02 0x02 c
```

Then:

```
i2cdetect -y 1
```

Should list detected devices- where presumably you should see the multiplexer and the VL53L1X.

lvermue, 2019-11-07:
@JimVanEeden Did you connect the VL53L1X to the right pins on the TCA9548A?
Based on your code that should be SDA1 and SCL1.
Your error message suggests that there is no VL53L1X connected to SDA1 and SCL1.

If this doesn't help, it would be helpful to know more about your setup. E.g. do you have the right pull-up resistors in place?

JimVanEeden, 2019-11-07:
Thanks guys, both suggestions help me out a lot. It turned out to be a hardware thing. Running `i2cset -y 1 0x70 0x01 0x02` first helped me get it working with /examples/distance.py, from there on I got distance_multiplexer.py working. 

Great library!

Gadgetoid, 2019-11-07:
I managed to secure a multiplexer from the shop to test with.

I'm not 100% sure it's actually working- since I managed to use `i2cset -y 1 0x70 0x02 c` to switch the multiplexer to channel 1 where I have my vl53l1x connected, and this causes the library to work... but if I switch the multiplexer to the wrong channel then the library does not seem to be changing it back.

Gadgetoid, 2019-11-07:
Okay- I found the problem.

The Python code for this library does a pure-python presence test for the VL53L1X on the i2c bus when it's initialised. This is before any of the C code is set up, and thus the new `i2c_multi` function never gets called. I've added some code in 5e365bb862597adaa37ef54b87cda2bb12ed4ebe to skip this check when a multiplexer is being used.

AlanWalker97, 2020-02-03:
> Okay- I found the problem.
> 
> The Python code for this library does a pure-python presence test for the VL53L1X on the i2c bus when it's initialised. This is before any of the C code is set up, and thus the new `i2c_multi` function never gets called. I've added some code in [5e365bb](https://github.com/pimoroni/vl53l1x-python/commit/5e365bb862597adaa37ef54b87cda2bb12ed4ebe) to skip this check when a multiplexer is being used.

Seriously this problem makes me sick and you solved this. Thank u so much :)

Gadgetoid, 2020-02-25:
This might be the most reactions I've ever got to a commit, and it was two lines of code 😆

closed 2020-02-25 by Gadgetoid

AlanWalker97, 2020-02-25:
Because i need that sensor and multiplexer for my thesis 😅

şunu yazdı: