pimoroni/vl53l1x-python #26: Negative Distance Value -1185

nrj127, 2019-11-22:
Hey,

I have been trying to test my code on a raspberrypi  (archlinux 64 architecture). However, I always get a negative distance value of -1185.  Also, while debugging the VL531LX.c file, I also noticed that the error code returned for both postive and negative values is 0. Has anyone faced a similar issue.

Thanks

nav2u, 2020-03-03:
Hello,

Similar issue here, different setup however.

I have 5x VL53L1X connected to a TCA9548A multiplexer on channels 0 through to 4. I  modified the Distance_multiplexer.py to read all 5 sensors with 100ms sleep between each read. Running on Raspbian Buster Lite, Raspberry Pi 3 B+, setup on a breadboard, facing the cealing (~1.6m from the breadboard).

Issue: Random sensors will randomly give me -1185mm readings or other random values and sometimes even just a 0mm reading.

Troubleshooting I've done so far: Tested with long, medium and short ranging options, increased the time between measurements, ran the program multiple times to ensure that all sensors at some stage or the other provided correct measurements (I was trying to check that the sensors aren't faulty), changed lighting conditions.

Looking for help/clues/ideas in figuring out whats going on. Thanks!

A few example measurements:

**Example 1**
```
Display the distance read from the sensor.
Press Ctrl+C to exit.

VL53L1X Start Ranging Address 0x29 TCA9548A Device 0 TCA9548A Address 0x70

VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID : 
ProductRevisionMajor : 1
ProductRevisionMinor : 15
VL53L1X Start Ranging Address 0x29 TCA9548A Device 1 TCA9548A Address 0x70

VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID : 
ProductRevisionMajor : 1
ProductRevisionMinor : 15
VL53L1X Start Ranging Address 0x29 TCA9548A Device 2 TCA9548A Address 0x70

VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID : 
ProductRevisionMajor : 1
ProductRevisionMinor : 15
VL53L1X Start Ranging Address 0x29 TCA9548A Device 3 TCA9548A Address 0x70

VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID : 
ProductRevisionMajor : 1
ProductRevisionMinor : 15
VL53L1X Start Ranging Address 0x29 TCA9548A Device 4 TCA9548A Address 0x70

VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID : 
ProductRevisionMajor : 1
ProductRevisionMinor : 15
Sensor 1 distance: 1578mm
Sensor 2 distance: 1116mm
Sensor 3 distance: 1634mm
Sensor 4 distance: 1574mm
Sensor 5 distance: 659mm
Sensor 1 distance: 1572mm
Sensor 2 distance: 1196mm
Sensor 3 distance: 1630mm
Sensor 4 distance: 1569mm
Sensor 5 distance: 741mm
Sensor 1 distance: 1565mm
[... 157 lines cut ...]
Sensor 3 distance: -1185mm
Sensor 4 distance: -1185mm
Sensor 5 distance: 1628mm
Sensor 1 distance: 1627mm
Sensor 2 distance: 1637mm
Sensor 3 distance: -1185mm
Sensor 4 distance: -1185mm
Sensor 5 distance: -1185mm
Sensor 1 distance: -1185mm
Sensor 2 distance: 1637mm
Sensor 3 distance: -1185mm
Sensor 4 distance: -1185mm
Sensor 5 distance: -1185mm
Sensor 1 distance: -1185mm
Sensor 2 distance: 1638mm
Sensor 3 distance: -1185mm
Sensor 4 distance: -1185mm
Sensor 5 distance: -1185mm
Sensor 1 distance: -1185mm
Sensor 2 distance: 1639mm
Sensor 3 distance: -1185mm
Sensor 4 distance: -1185mm
Sensor 5 distance: 1611mm

```

Gadgetoid, 2020-03-04:
The spurious negative readings look suspiciously like an overflow, but it's bizarre that they are always the same value. It could be a silent read error that's then being converted into a distance.

The measurement data struct which we're currently jus tpulling `RangeMilliMeter` from also hs a `RangeStatus` that's not being checked. It's possible in the case of these errant readings the `RangeStatus` is 1 (invalid) and the reading should be rejected.

The `get_distance` method in Python calls `getDistance` in the C bindings which is defined here:

https://github.com/pimoroni/vl53l1x-python/blob/22eaff440e601997750ad946d280bf2df4134201/python_lib/vl53l1x_python.c#L186-L195

airopti, 2020-05-10:
It seems that there is a problem in switching the channel of TCA9548a

mglowinski93, 2021-12-03:
I've similar issue with single device connected to I2C network:
[image]

@Gadgetoid is there any feedback on that?

soerenbrockmann, 2023-03-28:
For me the error value was gone when setting a timer `time.sleep(0.5).`

peter9477, 2025-02-23:
> The spurious negative readings look suspiciously like an overflow, but it's bizarre that they are always the same value. It could be a silent read error that's then being converted into a distance.
> 
> The measurement data struct which we're currently jus tpulling `RangeMilliMeter` from also hs a `RangeStatus` that's not being checked. It's possible in the case of these errant readings the `RangeStatus` is 1 (invalid) and the reading should be rejected.
> 
> The `get_distance` method in Python calls `getDistance` in the C bindings which is defined here:
> 
> [vl53l1x-python/python_lib/vl53l1x_python.c](https://github.com/pimoroni/vl53l1x-python/blob/22eaff440e601997750ad946d280bf2df4134201/python_lib/vl53l1x_python.c#L186-L195)
> 
> Lines 186 to 195 in [22eaff4](/pimoroni/vl53l1x-python/commit/22eaff440e601997750ad946d280bf2df4134201)
>  int32_t getDistance(VL53L1_Dev_t *dev) 
>  { 
>      VL53L1_Error Status = VL53L1_ERROR_NONE; 
>      int32_t current_distance = -1; 
>      Status = VL53L1_WaitMeasurementDataReady(dev); 
>      Status = VL53L1_GetRangingMeasurementData(dev, pRangingMeasurementData); 
>      current_distance = pRangingMeasurementData->RangeMilliMeter; 
>      VL53L1_ClearInterruptAndStartMeasurement(dev); 
>      return current_distance; 
>  }

That's hilariously bad code.  The supposed "-1" value that it will return in case of error is immediately overwritten with a supposed distance value (RangeMilliMeter).  (Do I even need to point out that nobody should use camel-case for "MilliMeter".)  Tracing further backwards, that value is set based on some other variable elsewhere which is supposed to have a distance.  The only way the return value can be -1185 is if something earlier stored an error value in some variable which then got processed as if it were an actual distance.  The true error values are all between -1 and -45 or so.  This code is a mess.