pimoroni/vl53l1x-python #27: Re: Can't get the VL53L1X to work on RPi 4

aylmer123, 2019-12-15:
Hi,

This is probably a simple issue, but I'm new to RPi and Python and trying to help my daughter with a school project - which the VL53L1X seemed ideal for. SO far I cannot get the sensor working with connections to SCL1 and SDA1 on a RPi 4 using Thonny IDE. Installed both the smbus2 and the vl53l1x libraries from here and starting with the most basic setup. I am getting the following errors that suggest there is something I'm not getting about the i2c bus - any helpful hints would be much appreciated: 
Traceback (most recent call last):
  File "/home/pi/Desktop/laserpi.py", line 10, in <module>
    tof = VL53L1X.VL53L1X(i2c_bus=1, i2c_address=0x29)
  File "/usr/local/lib/python3.7/dist-packages/VL53L1X.py", line 84, in __init__
    self._i2c = SMBus(i2c_bus)
  File "/usr/local/lib/python3.7/dist-packages/smbus2/smbus2.py", line 279, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.7/dist-packages/smbus2/smbus2.py", line 308, in open
    self.fd = os.open(filepath, os.O_RDWR)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'

Gadgetoid, 2019-12-16:
The sailient part of the error in this case is `FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'` which means the code cannot find the device node at `/dev/i2c-1`. More often than not this means i2c just needs enabling (it's turned off by default) which you can do either via the Raspberry Pi Configuration utility (found in the Raspberry Pi menu) or by adding `dtoverlay=i2c=on` to the `/boot/config.txt` file.

aylmer123, 2019-12-17:
Tx Gadgetoid, that fixed the problem - I was going through all the lines of code trying to think what could be wrong...  and missed the bleeding obvious. Enabled i2c and the code worked great.
On another issue - having played with the sensor it seems to have a flaw at short distances. For example, at 5 cm the first two measurements are low (often 0 mm for the first, 40 mm for the second, and then precise and accurate from then on). At 30 cm the first measurement is low (generally ~280 mm) and then its accurate and precise from then on. At 60 cm the sensor is accurate and precise from the first measurement. The behaviour is not impacted by the measurement length time, the ranging setting (i.e. 1,2 or 3) or the amount of usage. Since the behaviour is repetitive but only at short distances I think it must be the sensor (a Pololu break-out board) rather than the code. Has anyone else seen this behaviour?
Tx for any thoughts.

Gadgetoid, 2019-12-18:
Spurious readings are not uncommon when it comes to sensors, and usually you'd add code to reject them.

It looks like in the case of the VL53L1X we might have to use the `RangeStatus` register to ensure readings are valid.

There's some - possibly related - discussion here: https://community.st.com/s/question/0D50X00009sUiJUSA0/out-of-range-readings-of-vl53l1x

closed 2020-08-03 by Gadgetoid