pimoroni/vl53l1x-python #30: Roi module error help please

Retry3, 2020-01-21:
I am student and low level programming language, roi.py file run and 
--------------------------------------------------------------------
Traceback (most recent call last):
  File "./roi.py", line 61, in <module>
    roi = scan("default")
  File "./roi.py", line 56, in scan
    return VL53L1X.VL53L1xUserRoi(0, 15, 15, 0)
AttributeError: 'module' object has no attribute 'VL53L1xUserRoi'
-------------------------------------------------------------------------
help please thank you.

alex-savin, 2020-01-29:
Have the same issue
```
Traceback (most recent call last):
  File "roi.py", line 61, in <module>
    roi = scan("default")
  File "roi.py", line 56, in scan
    return VL53L1X.VL53L1xUserRoi(0, 15, 15, 0)
AttributeError: module 'VL53L1X' has no attribute 'VL53L1xUserRoi'
```

Gadgetoid, 2020-03-11:
I suspect this is because I haven't released a new version of the VL53L1X library that includes the ROI functionality from https://github.com/pimoroni/vl53l1x-python/pull/29 - in fact that's certainly the case: https://github.com/pimoroni/vl53l1x-python/compare/v0.0.4...master

In the interim you should be able to install the library from the cloned Git repository:

```
python3 setup.py build
sudo python3 setup.py install
```

Gadgetoid, 2020-03-11:
`sudo pip3 install vl53l1x --upgrade` or `sudo pip install vl53l1x --upgrade`  should now work.

closed 2020-03-11 by Gadgetoid