pimoroni/vl53l1x-python #33: Multiple VL53L1X on RPi with software I2C not working

nav2u, 2020-03-16:
Hello,

I've been trying to get multiple VL53L1X to work with a Raspberry pi. I've used the TCA9548A multiplexer but had some issues (see issue #26) and so decided to try using software I2C on the Raspberry Pi as an alternate.

When using the software I2C with multiple sensors, it seems like distance is only read from one sensor and that distance is then displayed as results for all the other sensor running on different I2C busses.

Some testing I did:
- If I use a single I2C bus, I get data from the correct sensor. Tested this on all buses individually (1 through to 6, excluding bus 2). I think this confirms that all the sensors are working and that the software I2C bus has been enabled correctly.
- Ran two sensors together on buses 1 and 3. Distance results for both sensors were being read from sensor on bus 3. *
- Ran the above test again but initialised/opened sensor on bus 3 first and then bus 1. Distance results for both sensors were being read from sensor on bus 1. *
- Ran all five sensors together on buses 1, 3, 4, 5 and 6. Distance results for all sensors were being read from a single sensor on bus 6. *
- Ran the above test again but switched around the sensor initialisation in the following bus order 1, 3, 4, 6, 5. Distance results for all sensors were being read from sensor on bus 5. *

( *Confirmed by physically blocking individual sensors )

So it seems like the sensor on the last initialised/opened bus is always used by the get_distance function (I'm not sure how to explain whats going on here)? Any ideas what might be going on here? Perhaps I've missed something?

Code I used for testing:

```
import time
import sys
import signal
import VL53L1X

tof1 = VL53L1X.VL53L1X(i2c_bus=1, i2c_address=0x29)
tof2 = VL53L1X.VL53L1X(i2c_bus=3, i2c_address=0x29)
tof3 = VL53L1X.VL53L1X(i2c_bus=4, i2c_address=0x29)
tof4 = VL53L1X.VL53L1X(i2c_bus=5, i2c_address=0x29)
tof5 = VL53L1X.VL53L1X(i2c_bus=6, i2c_address=0x29)

time.sleep(1)
tof1.open()
tof1.start_ranging(3)
tof2.open()
tof2.start_ranging(3)
tof3.open()
tof3.start_ranging(3)
tof4.open()
tof4.start_ranging(3)
tof5.open()
tof5.start_ranging(3)

def exit_handler(signal, frame):
    global running
    running = False
    tof1.stop_ranging()
    tof2.stop_ranging()
    tof3.stop_ranging()
    tof4.stop_ranging()
    tof5.stop_ranging()
    print()
    sys.exit(0)

running = True
signal.signal(signal.SIGINT, exit_handler)

while running:
    distance_in_mm = tof1.get_distance()
    print("Sensor 1 distance: {}mm".format(distance_in_mm))
    distance_in_mm = tof2.get_distance()
    print("Sensor 2 distance: {}mm".format(distance_in_mm))
    time.sleep(0.5)
    distance_in_mm = tof3.get_distance()
    print("Sensor 3 distance: {}mm".format(distance_in_mm))
    distance_in_mm = tof4.get_distance()
    print("Sensor 4 distance: {}mm".format(distance_in_mm))
    time.sleep(0.5)
    distance_in_mm = tof5.get_distance()
    print("Sensor 5 distance: {}mm".format(distance_in_mm))
```

Gadgetoid, 2020-03-16:
I'm willing to bet- although I have no hardware with me right now to perform any testing- that the handing of Python i2c functions into the C library is what's causing this problem. See:

https://github.com/pimoroni/vl53l1x-python/blob/a0418c59607d408ae3c94a1d1d45b7a8e5e1a0d8/python/VL53L1X.py#L116-L159

The most relevant lines are:

https://github.com/pimoroni/vl53l1x-python/blob/a0418c59607d408ae3c94a1d1d45b7a8e5e1a0d8/python/VL53L1X.py#L155-L159

While each instance of the VL53L1X class gets its own instance of the `_TOF_LIBRARY.initialise()` return value, ie a `VL53L1_DEV` object, the call to set i2c functions is *global*, setting the i2c read/write/muilti pointers for *all* instances:

https://github.com/pimoroni/vl53l1x-python/blob/505fd6be452f5e9172fa833da1be3b8e10c5b5eb/api/platform/vl53l1_platform.c#L117-L135

## The Fix

As near as I can tell, nothing calls these `i2c_write` or `i2c_read` functions directly. They are handled via wrappers that receive an instance of the `VL53L1_DEV` object:

https://github.com/pimoroni/vl53l1x-python/blob/505fd6be452f5e9172fa833da1be3b8e10c5b5eb/api/platform/vl53l1_platform.c#L225-L245

The *right* solultion, I believe, is therefore to move the `i2c_multi_func`, `i2c_read_func` and `i2c_write_func` function pointers out of the global namespace, and instead encapsulate them into `VL523L1X` here:

https://github.com/pimoroni/vl53l1x-python/blob/14bbf3aa975f9193716efb4743a6b33e92c64d1f/api/platform/vl53l1_platform_user_data.h#L78-L92

Now I'm *thoroughly terrible* with pointers, but I may be able to get this up and running blind and push it into a branch for you to test.

I'd been suggesting using multiple i2c channels without fully appreciating how broken the current library setup would make that. But it's easily fixable by the looks of it.

Gadgetoid, 2020-03-16:
Okay, made a first attempt. I'll be surprised if I haven't missed something, but fingers crossed!

See: https://github.com/pimoroni/vl53l1x-python/pull/34

Gadgetoid, 2025-01-08:
Discussion should probably continue at #34 (though it's been a couple of years...)

closed 2025-01-08 by Gadgetoid