pimoroni/vl53l1x-python #37: Segmentation fault when calling start_ranging

slishak, 2020-05-03:
I have a brand new Pimoroni VL53L1X connected to the I2C bus on my Raspberry Pi 4B. Using this Python library, I get a segmentation fault when I start ranging:

```
Python 2.7.17 (default, Apr 15 2020, 17:20:14)
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import VL53L1X
>>> tof = VL53L1X.VL53L1X()
>>> tof.open()
VL53L1X Start Ranging Address 0x29

VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID :
ProductRevisionMajor : 1
ProductRevisionMinor : 11
>>> tof.start_ranging(1)
Segmentation fault
```

I've tried building from source and got the same error. Also the same on Ubuntu instead of Raspbian.

I thought I just had a dodgy sensor, but if I use the [qwiic_vl53l1x ](https://qwiic-vl53l1x-py.readthedocs.io/en/latest/)library from SparkFun, everything works fine:

```
Python 2.7.17 (default, Apr 15 2020, 17:20:14)
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import qwiic_vl53l1x
>>> tof = qwiic_vl53l1x.QwiicVL53L1X()
>>> tof.sensor_init()
>>> tof.set_distance_mode(1)
1
>>> tof.start_ranging()
>>> tof.get_distance()
190
>>> tof.stop_ranging()
```

Gadgetoid, 2020-07-31:
Sorry, I seem to have missed this- did you get any closer to figuring out what the problem might be?

I can only guess the device was responding in an unexpected way, such that it caused the library to crash, but I've no idea quite how.

slishak, 2020-08-04:
This issue came from a lockdown project that I'm no longer working on as I'm back to work, so unfortunately don't really have the time to continue debugging now, sorry! In the end I just continued with the qwiic library instead; I assume the reason this library doesn't work for me is just a peculiarity with my device as nobody else seems to have the same issue.

If/when I do get back to this, what's the best way for me to debug the issue?

edahle2022, 2021-08-10:
I'm having the same exact problem with the sensor on Ubuntu using Python. If anyone's figured out the solution, I'd appreciate it.

eladpar-rnd, 2021-08-17:
having the same problem Rasberry pi 4 ubuntu 18.04

eladpar-rnd, 2021-08-18:
after some debugging I've identified the problem.
the address allocated in the c file vl53l1x_python.c VL53L1_DEV * dev
cannot be accessed again in all functions of c file (_TOF_LIBRARY) 
it doesn't matter what address i allocate memory (tried with int * ) after sending it to the python file and trying to accesses again with the c file gets a seg fault

eladpar-rnd, 2021-08-18:
another comment:
turns out it happens in Python 2.7.15+ and Python 3.7.11 (that I checked)
but in Python 3.6.8 when pip installing it works fine

Dko1905, 2022-10-16:
Same issue here with a Raspberry Pi 4 running Raspberry OS bullseye using Python 3.9.2 installed from source.

sibujacob, 2023-09-03:
Hello,

Is there anyone who has recovered from this fault "  segmentation fault (core dumped)   "  error  while using this code on a RasPi 3b+ .  i am currently using the ubuntu os on raspi

Darth-Carrotpie, 2024-09-01:
Same fault.
Using RPI5

Odinroast, 2026-01-21:
Just made a pr with the fix that I found for this issue! 

https://github.com/Odinroast/vl53l1x-python/tree/fix-segfault-ctypes-declarations

Blaise-wig, 2026-03-19:
I have the same Segmentation fault with Jetson Nano. My Jetson Nano has ROS Melodic installed with Python 2.7

Gadgetoid, 2026-08-25:
This should have been fixed by #60 which I've merged into #62 pending a final merge and release.

*Thank you* @Odinroast

closed 2026-08-25 by Gadgetoid