pimoroni/vl53l1x-python #40: Use multiple VL53L1X sensort with I2C multiplexer from 8086.net

pdandis, 2020-07-05:
Hi,

I’m using a I2C multiplexer from 8086-X, because the VL53L1X sensor are not able to change the i2c address x29 permanently.
I2Cdetect shows all i2c busses as expected and I assumed, that my code should run. I modified an example from your github lib. To avoid pyhsical wiring issues, I change all cabling etc.

#!/usr/bin/env python

import time
import sys
import signal

import VL53L1X

print("""distance.py

Display the distance read from the sensor.

Uses the "Short Range" timing budget by default.

Press Ctrl+C to exit.

""")

tof1 = VL53L1X.VL53L1X(i2c_bus=7, i2c_address=0x29)
tof1.open()
tof2 = VL53L1X.VL53L1X(i2c_bus=8, i2c_address=0x29)
tof2.open()

tof1.start_ranging(1)  # Start ranging
                      # 0 = Unchanged
                      # 1 = Short Range
                      # 2 = Medium Range
                      # 3 = Long Range

tof2.start_ranging(1)

running = True

def exit_handler(signal, frame):
    global running
    running = False
    tof1.stop_ranging()
    tof2.stop_ranging()
    print()
    sys.exit(0)

while running:
    distance_in_mm1 = tof1.get_distance()
    print("Distance ToF1: {}mm".format(distance_in_mm1))
    time.sleep(5)
    distance_in_mm2 = tof2.get_distance()
    print("Distance ToF2: {}mm".format(distance_in_mm2))
    time.sleep(5)

I got no errors, but the the measurment of the two ToF sensors are most wrong. Massively to short, sometimes ToV2 show the same values as ToV1. I assume that the seperation of the two i2c-busses are not working  probably. Any idea how I could sove this problem?
Many thx in advance.

Peter

Gadgetoid, 2020-07-31:
I wonder if some quirk in the way that i2c is multiplexed is fundamentally incompatible with this library.

At the Pi end, all of these devices are sharing the same physical bus, but at the other end they are routed via the TCA9548A using an overlay that- presumably- muxes them when communication with the relevant i2c bus is detected.

Maybe the best way would be to drop the dtoverlay supplied by 8086 and try the native support for the TCA9548A built into this library that's demonstrated in this example: https://github.com/pimoroni/vl53l1x-python/blob/master/examples/distance_multiplexer.py

pdandis, 2020-08-01:
my thx I will checked and give feedback the next days.

pdandis, 2020-08-02:
Hi,

I just checked the /boot/config.txt ..... dtoverlay was not enabled ...... during my tests.

Gadgetoid, 2020-08-03:
🤦 
It happens to us all.
Does it work with it enabled?

thirdr, 2025-01-08:
Closing this to clear our decks for the new year, if it's still troubling you please re-open! :)

closed 2025-01-08 by thirdr