pimoroni/vl53l1x-python #41: Error Installing VL53L1X

Ryan-W96, 2020-07-08:
Hi there, 

I am new programming and trying to use this sensor as part of a uni project. 

pip install smbus2 was successful, vl53l1x was not (below). 

C:\Users\ryan>pip install vl53l1x
Collecting vl53l1x
  Using cached VL53L1X-0.0.5.tar.gz (182 kB)
Requirement already satisfied: smbus2 in c:\users\ryan\appdata\local\programs\python\python38-32\lib\site-packages (from vl53l1x) (0.3.0)
Building wheels for collected packages: vl53l1x
  Building wheel for vl53l1x (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\users\ryan\appdata\local\programs\python\python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\ryan\\AppData\\Local\\Temp\\pip-install-x683p1_g\\vl53l1x\\setup.py'"'"'; __file__='"'"'C:\\Users\\ryan\\AppData\\Local\\Temp\\pip-install-x683p1_g\\vl53l1x\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\ryan\AppData\Local\Temp\pip-wheel-tonndecq'
       cwd: C:\Users\ryan\AppData\Local\Temp\pip-install-x683p1_g\vl53l1x\
  Complete output (84 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win32-3.8
  copying python\VL53L1X.py -> build\lib.win32-3.8
  running build_ext
  building 'vl53l1x_python' extension
  creating build\temp.win32-3.8
  creating build\temp.win32-3.8\Release
  creating build\temp.win32-3.8\Release\api
  creating build\temp.win32-3.8\Release\api\core
  creating build\temp.win32-3.8\Release\api\platform
  creating build\temp.win32-3.8\Release\python_lib
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -Iapi/core -Iapi/platform -Ic:\users\ryan\appdata\local\programs\python\python38-32\include -Ic:\users\ryan\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tcapi/core/vl53l1_api_calibration.c /Fobuild\temp.win32-3.8\Release\api/core/vl53l1_api_calibration.obj -std=c99
  cl : Command line warning D9002 : ignoring unknown option '-std=c99'
  vl53l1_api_calibration.c
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -Iapi/core -Iapi/platform -Ic:\users\ryan\appdata\local\programs\python\python38-32\include -Ic:\users\ryan\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tcapi/core/vl53l1_core.c /Fobuild\temp.win32-3.8\Release\api/core/vl53l1_core.obj -std=c99
  cl : Command line warning D9002 : ignoring unknown option '-std=c99'
  vl53l1_core.c
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -Iapi/core -Iapi/platform -Ic:\users\ryan\appdata\local\programs\python\python38-32\include -Ic:\users\ryan\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tcapi/core/vl53l1_core_support.c /Fobuild\temp.win32-3.8\Release\api/core/vl53l1_core_support.obj -std=c99
  cl : Command line warning D9002 : ignoring unknown option '-std=c99'
  vl53l1_core_support.c
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -Iapi/core -Iapi/platform -Ic:\users\ryan\appdata\local\programs\python\python38-32\include -Ic:\users\ryan\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tcapi/core/vl53l1_api_core.c /Fobuild\temp.win32-3.8\Release\api/core/vl53l1_api_core.obj -std=c99
  cl : Command line warning D9002 : ignoring unknown option '-std=c99'
  vl53l1_api_core.c
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -Iapi/core -Iapi/platform -Ic:\users\ryan\appdata\local\programs\python\python38-32\include -Ic:\users\ryan\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tcapi/core/vl53l1_api_preset_modes.c /Fobuild\temp.win32-3.8\Release\api/core/vl53l1_api_preset_modes.obj -std=c99
  cl : Command line warning D9002 : ignoring unknown option '-std=c99'
  vl53l1_api_preset_modes.c
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -Iapi/core -Iapi/platform -Ic:\users\ryan\appdata\local\programs\python\python38-32\include -Ic:\users\ryan\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tcapi/core/vl53l1_silicon_core.c /Fobuild\temp.win32-3.8\Release\api/core/vl53l1_silicon_core.obj -std=c99
  cl : Command line warning D9002 : ignoring unknown option '-std=c99'
  vl53l1_silicon_core.c
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -Iapi/core -Iapi/platform -Ic:\users\ryan\appdata\local\programs\python\python38-32\include -Ic:\users\ryan\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\in [... cut ...] )(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\ryan\AppData\Local\Temp\pip-record-df_u6uit\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\ryan\appdata\local\programs\python\python38-32\Include\vl53l1x' Check the logs for full command output.

Sorry for so many lines but this is what I'm getting and don't know where to go from here. 
Many thanks,
Ryan

Gadgetoid, 2020-07-08:
I'm guessing you're trying to install this library on your development machine so you can use intellisense to write code against it?

Unfortunately that wont work.

I've tried exactly this with other libraries targeted at Linux/Raspberry Pi- in theory it should be possible to get the intellisense features by some other means, but this library (and most others) will never be possible to install on Windows.

If you're using something like Visual Studio Code then you should have a Raspberry Pi set up and running with this sensor connected, and use remote development to edit and run files *on* your Pi with its own Python environment. See: https://code.visualstudio.com/docs/remote/remote-overview

Ryan-W96, 2020-07-08:
I was and okay, that makes sense. 

Great, I'll look into that, thanks for your quick response! And thank you for your help.

Gadgetoid, 2020-07-12:
No worries! Hope you get up and running.

closed 2020-07-12 by Gadgetoid