pimoroni/vl53l1x-python #42: Cannot install vl53l1x via pip

trojek, 2020-08-24:
I use RPi 3 and I have python (both 2 and 3 version) and pip installed.

When I execute:

```pip install vl53l1x```

I got:

```Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting vl53l1x
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://www.piwheels.org/simple/vl53l1x/
```

How can I install vl541x package?

Gadgetoid, 2020-08-24:
It's on PyPI so presumably your pip is not failing over from piwheels.org to PyPI... I don't know why this might be.

https://pypi.org/project/VL53L1X/#history

Perhaps this will work?:

```
pip install -i https://pypi.org/simple/ VL53L1X
```

Gadgetoid, 2020-08-24:
Oh it looks like PiWheels is case-sensitive and doesn't do the redirect that PyPI does. Some update to either, or pip, has potentially totally broken the ability to install this package on a Pi.

I've raised an issue about this here: https://github.com/piwheels/packages/issues/145

In the mean time you *could* grab the appropriate .whl from here: https://pypi.org/project/VL53L1X/#files

And just supply the filename to the `pip install` command.

trojek, 2020-08-24:
I tried `pip install -i https://pypi.org/simple/ VL53L1X` but I got the same error.
I also try to execute this command on my laptop with Fedora installed and it works.

On raspberry pi I downloaded .whl file (`wget https://files.pythonhosted.org/packages/df/05/7aff24243ae12ac1b8b1f1721e9ee6460be7373f71cce795fe2b79bae2d7/VL53L1X-0.0.5-cp37-cp37m-linux_armv7l.whl`) and installed as you suggested (`pip3 install VL53L1X-0.0.5-cp37-cp37m-linux_armv7l.whl`)

Everything works, thank you.

closed 2020-08-24 by trojek