pimoroni/vl53l1x-python #48: does not measure

almirus, 2021-09-25:
i have vl53lxx-v2
[image]
after start get:

> Display the distance read from the sensor.
> 
> Uses the "Short Range" timing budget by default.
> 
> Press Ctrl+C to exit.
> 
> 
> VL53L1X Start Ranging Address 0x29
> 
> VL53L0X_GetDeviceInfo:
> Device Name : VL53L1 cut1.0
> Device Type : VL53L1
> Device ID :
> ProductRevisionMajor : 1
> ProductRevisionMinor : 0
> Distance: 2011mm
> Distance: 2011mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
> Distance: 0mm
>

IngramINhaidian, 2022-05-06:
Hi, did you solve this problem? I am in the same situation with only 2011 and 0mm as output. I suppose that this is some kind of hardware problem? Looking forward to answer.

almirus, 2022-05-06:
@IngramINhaidian if I remember correctly, I used a different library. This one is not suitable.

IngramINhaidian, 2022-05-07:
@almirus Can you remember what library it was? It's fine if it's too hard to remember. Anyway, thanks!

motey, 2022-06-27:
Got the same issue with my sensor (https://de.aliexpress.com/item/4000501203735.html)
Same output (starts with 2011mm and after that 0mm)
any new ideas here?

almirus, 2022-06-27:
@IngramINhaidian @motey try https://github.com/pimoroni/VL53L0X-python

almirus, 2022-07-17:
https://github.com/adafruit/Adafruit_CircuitPython_VL53L0X

closed 2025-01-08 by thirdr