pimoroni/vl53l1x-python #49: Reading does not return to expected value after measuring shorter distance

fuchrist, 2021-09-27:
Hi all,

I just try to hook up a VL53L1X to a Raspberry Pi and it initially worked very well.

However I encountered a problem (see images):

Once I start a measurement (e.g. every second), I seem to get reasonable readings.

When I then put my hand into the beam, I also seem to measure the correct distance.

However, when I pull my hand back out, the reading doesn't go back to the intial value.

--> Could somebody help me out here? Any hints on why this is happening? Could this e.g. be caused by a missing pull-up resistor?

When I close() and open() the sensor again, everything is "back to normal"...

[image]
[image]

Gadgetoid, 2021-09-29:
Accuracy/repeatability varies by distance, ambient light and a bunch of other factors, depends upon long/short distance mode, etc.

I'd recommend looking over sections 3.4, 3.5 and 3.6 of the datasheet to get an idea of what to expect - https://www.st.com/resource/en/datasheet/vl53l1x.pdf 

There should probably be a way to get the measurement error code, since many of them shed light on configuration/environmental errors. This guide mentions some range statuses which I think are converted to error codes by this library: https://www.st.com/resource/en/user_manual/um2510-a-guide-to-using-the-vl53l1x-ultra-lite-driver-stmicroelectronics.pdf

> There are five range statuses: 0, 1, 2, 4, and 7. When the range status is 0, there is no error. Range status 1 and
2 are error warnings while range status 4 and 7 are errors.
When the range status is 1, there is a sigma failure. This means that the repeatability or standard deviation of
the measurement is bad due to a decreasing signal noise ratio. Increasing the timing budget can improve the
standard deviation and avoid a range status 1.
When the range status is 2, there is a signal failure. This means that the return signal is too week to return a
good answer. The reason is because the target is too far, or the target is not reflective enough, or the target is too
small. Increasing the timing buget might help, but there may simply be no target available.
When the range status is 4, the sensor is "out of bounds". This means that the sensor is ranging in a “non-
appropriated” zone and the measured result may be inconsistent. This status is considered as a warning but, in
general, it happens when a target is at the maximum distance possible from the sensor, i.e. around 5 m. However,
this is only for very bright targets.
Range status 7 is called "wraparound". This situation may occur when the target is very reflective and the
distance to the target/sensor is longer than the physical limited distance measurable by the sensor. Such
distances include approximately 5 m when the senor is in Long distance mode and approximately 1.3 m when the
sensor is in Short distance mode. Example: a traffic sign located at 6 m can be seen by the sensor and returns a
range of 1 m. This is due to “radar aliasing”: if only an approximate distance is required, we may add 6 m to the
distance returned. However, that is a very approximate estimation.

But ultimately it's hard to pin down why this might be happening

Does the signal return to the previous value if you leave it long enough. IE: does it look like sample averaging? (from your paste, it doesn't seem so)

thirdr, 2025-01-08:
Closed due to no further activity. If it's still troubling you please re-open! :)

closed 2025-01-08 by thirdr