pimoroni/vl53l1x-python #51: Should the library be built independently and not via pip tool?

mglowinski93, 2021-12-03:
Hi,
I've a question about proper installation of this library.

For the first attempt I installed it by `pip install vl53l1x==0.0.5`.
In this case, I noticed that debug information appears (see https://github.com/pimoroni/vl53l1x-python/issues/47) and that exceptions will not be raised when the device is not active (see below code snippet):

```python
OSError: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 232, in 'calling callback function'
  File "/home/pi/.local/share/virtualenvs/PlugsTester-EektgUbj/lib/python3.7/site-packages/VL53L1X.py", line 142, in _i2c_write
    self._i2c.i2c_rdwr(msg_w)
  File "/home/pi/.local/share/virtualenvs/PlugsTester-EektgUbj/lib/python3.7/site-packages/smbus2/smbus2.py", line 658, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 121] Remote I/O erro
VL53L1X Start Ranging Address 0x29

VL53L0X_GetDeviceInfo:
Device Name : VL53L1 cut1.1
Device Type : VL53L1
Device ID : 
ProductRevisionMajor : 1
ProductRevisionMinor : 15

Distance: 100 cm
```

That is something what I would like to avoid, in other words to fail the script when device is not available.

So I decided to build a package on my own. I've downloaded this repository and executed `python setup.py build`.
I can see a library `vl53l1x_python.so` file inside `build` folder.
Once I move [VL53L1X.py](https://github.com/pimoroni/vl53l1x-python/blob/master/python/VL53L1X.py) and [distance.py](https://github.com/pimoroni/vl53l1x-python/blob/master/examples/distance.py) to build folder I can successfully communicate with the sensor and `DEBUG` output is not printed. I guess because I've build it without `-DDEBUG` flad as mentioned in https://github.com/pimoroni/vl53l1x-python/pull/39.

Nevertheless, I still can't get the script failing when device is not available.
So I've two questions: 
1) should i rather build package on my own or install it with pip?
2) How to make script fail when there is an exception?

Gadgetoid, 2022-02-10:
Looks like I need to publish a new release via pip, v0.0.5 doesn't include the latest changes.

The error you want to catch when a sensor is unplugged or unavailable is probably `OSError`:

```
Traceback (most recent call last):
  File "/home/pi/Documents/vl53l1x-python/python/VL53L1X.py", line 145, in _i2c_write
    self._i2c.i2c_rdwr(msg_w)
  File "/usr/local/lib/python3.9/dist-packages/smbus2/smbus2.py", line 658, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 121] Remote I/O error
Distance: -1mm
```

mglowinski93, 2022-02-10:
I've already found workaround :) 
But thanks anyway, good that you aware about this issue.

closed 2026-08-25 by Gadgetoid